Lienol/openwrt #571: HC5962 （B70）无限重启请教

dwj0, 2021-04-18:
你好，我编译了HC5962的固件，但从Breed中更新到路由器中后无限重启。
我之前用的是编译自LEAN的固件，这个固件一切正常

我观察到您的固件和LEAN固件对闪存分区的分配是不一样的，可能是这个原因不从的LEAN的固件中直接升级。因此我先在Breed中升级openwrt-ramips-mt7621-hc5962-initramfs-kernel.bin，升级完这个是能进路由器的，然后我在路由器中升级openwrt-ramips-mt7621-hc5962-squashfs-sysupgrade.bin后就无限重启了。

我发现编译的时候没有生成openwrt-ramips-mt7621-hc5962-squashfs-factory.bin固件，我想是不是需要先升级到这个固件才行。
所以也请教一下生成openwrt-ramips-mt7621-hc5962-squashfs-factory.bin需要更改哪里

dwj0, 2021-04-18:
我想我找到原因了
我把固件尺寸减小到25M之内，就出现了文件openwrt-ramips-mt7621-hc5962-squashfs-factory.bin
并且也能够正常使用了，但软件包中显示可用空间只有2M，因该是之前的固件大小为31M超过了一定大小才不停重启。
可我的FLASH是128M啊。。。

Lienol, 2021-04-18:
因为我现在已经没有任何硬路由设备，可以参考
https://github.com/coolsnowwolf/lede/issues/5673
https://github.com/coolsnowwolf/lede/issues/5813
成功后可以发个PR

closed 2021-04-25 by dwj0