Lienol/openwrt #577: 希望集成bypass插件

lon91ong, 2021-04-25:
项目地址: [bypass](https://github.com/garypang13/luci-app-bypass)
测试过[多个插件](https://www.right.com.cn/forum/thread-4104202-1-1.html)之后, 总是这样那样的问题, 最后发现bypass这个插件简单实用

就是原作者garypang的OpenWrt只有5.4内核支持, 在低配置的机器(mt7620a方案)上跑5.4内核实在是小马拉大车...

自己[折腾了一番](https://github.com/lon91ong/Actions-OpenWrt/tree/lienol-19.07), 虽然也编译成功了, 但似乎有"暗病", 经常早上重启一段时间后, 路由器会"假死", 连不上外网, 查看系统日志, 只有upnp有几行报错的:
```
Sun Apr 25 11:19:31 2021 daemon.err miniupnpd[2868]: upnp_event_recv: recv(): Connection reset by peer
Sun Apr 25 11:19:31 2021 daemon.err miniupnpd[2868]: upnpevents_processfds: 0x429890, remove subscriber uuid:d0674efe-7527-44cb-8101-2a5569a5b279 after an ERROR cb: http://192.168.77.220:2869/upnp/eventing/lerllbnmqb
Sun Apr 25 11:19:31 2021 daemon.err miniupnpd[2868]: upnp_event_recv: recv(): Connection reset by peer
Sun Apr 25 11:19:31 2021 daemon.err miniupnpd[2868]: upnpevents_processfds: 0x4298d0, remove subscriber uuid:c870809a-7e88-4872-9156-b6188c2ac7b6 after an ERROR cb: http://192.168.77.220:2869/upnp/eventing/whhmvrjlhn
......
Sun Apr 25 11:19:49 2021 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED bc:96:80:3d:73:40
Sun Apr 25 11:19:49 2021 daemon.info hostapd: wlan0: STA bc:96:80:3d:73:40 IEEE 802.11: disassociated
Sun Apr 25 11:19:53 2021 daemon.info hostapd: wlan0: STA bc:96:80:3d:73:40 IEEE 802.11: authenticated
Sun Apr 25 11:19:53 2021 daemon.info hostapd: wlan0: STA bc:96:80:3d:73:40 IEEE 802.11: associated (aid 2)
Sun Apr 25 11:19:53 2021 daemon.notice hostapd: wlan0: AP-STA-CONNECTED bc:96:80:3d:73:40
Sun Apr 25 11:19:53 2021 daemon.info hostapd: wlan0: STA bc:96:80:3d:73:40 WPA: pairwise key handshake completed (RSN)
```
再过一会儿就又好了,  还是自己水平不够, 希望L大能出手

另外[garypang13/turboacc](https://github.com/garypang13/openwrt-packages/tree/master/luci-app-turboacc)插件比
`luci-app-sfe`多支持**硬件**流量分载, 希望也能够支持

OPnb, 2021-04-28:
Li大不参与爬墙插件开发了已经。passwall都转给其他人own了。你这还拿个不声明参考源码的抄袭货 bypass 之流要加到这里。。。。。。

Lienol, 2021-04-28:
Where my live, there is no need to use such plugins.

closed 2021-04-28 by Lienol