Lienol/openwrt #585: PPTP 服务器无法连接，出错日志如下。去年编译的老版本是可以连接的

shanzigang, 2021-05-07:
Fri May  7 22:30:35 2021 daemon.info pptpd[13165]: CTRL: Client 49.78.205.197 control connection started
Fri May  7 22:30:35 2021 daemon.info pptpd[13165]: CTRL: Starting call (launching pppd, opening GRE)
Fri May  7 22:30:35 2021 daemon.notice pppd[13166]: pppd 2.4.9 started by root, uid 0
Fri May  7 22:30:35 2021 daemon.info pppd[13166]: Using interface ppp2
Fri May  7 22:30:35 2021 daemon.notice pppd[13166]: Connect: ppp2 <--> /dev/pts/0
Fri May  7 22:30:38 2021 daemon.err uhttpd[3213]: Model pptpd/online returns SimpleForm but is dispatched via cbi(),
Fri May  7 22:30:38 2021 daemon.err uhttpd[3213]: please change admin/vpn/pptpd/online to use the form() action instead.
Fri May  7 22:30:49 2021 daemon.err uhttpd[3213]: Model pptpd/online returns SimpleForm but is dispatched via cbi(),
Fri May  7 22:30:49 2021 daemon.err uhttpd[3213]: please change admin/vpn/pptpd/online to use the form() action instead.
Fri May  7 22:30:52 2021 daemon.err uhttpd[3213]: Model pptpd/online returns SimpleForm but is dispatched via cbi(),
Fri May  7 22:30:52 2021 daemon.err uhttpd[3213]: please change admin/vpn/pptpd/online to use the form() action instead.
Fri May  7 22:30:55 2021 daemon.err uhttpd[3213]: Model pptpd/online returns SimpleForm but is dispatched via cbi(),
Fri May  7 22:30:55 2021 daemon.err uhttpd[3213]: please change admin/vpn/pptpd/online to use the form() action instead.
Fri May  7 22:30:58 2021 daemon.err uhttpd[3213]: Model pptpd/online returns SimpleForm but is dispatched via cbi(),
Fri May  7 22:30:58 2021 daemon.err uhttpd[3213]: please change admin/vpn/pptpd/online to use the form() action instead.
Fri May  7 22:31:00 2021 daemon.err uhttpd[3213]: Model pptpd/online returns SimpleForm but is dispatched via cbi(),
Fri May  7 22:31:00 2021 daemon.err uhttpd[3213]: please change admin/vpn/pptpd/online to use the form() action instead.
Fri May  7 22:31:03 2021 daemon.err uhttpd[3213]: Model pptpd/online returns SimpleForm but is dispatched via cbi(),
Fri May  7 22:31:03 2021 daemon.err uhttpd[3213]: please change admin/vpn/pptpd/online to use the form() action instead.
Fri May  7 22:31:05 2021 daemon.warn pppd[13166]: LCP: timeout sending Config-Requests
Fri May  7 22:31:05 2021 daemon.notice pppd[13166]: Connection terminated.
Fri May  7 22:31:05 2021 daemon.info : 07[KNL] interface ppp2 deleted
Fri May  7 22:31:05 2021 daemon.notice pppd[13166]: Modem hangup
Fri May  7 22:31:05 2021 daemon.info pppd[13166]: Exit.
Fri May  7 22:31:05 2021 daemon.err pptpd[13165]: GRE: read(fd=6,buffer=413500,len=8196) from PTY failed: status = -1 error = I/O error, usually caused by unexpected termination of pppd, check option syntax and pppd logs
Fri May  7 22:31:05 2021 daemon.err pptpd[13165]: CTRL: PTY read or GRE write failed (pty,gre)=(6,7)
Fri May  7 22:31:05 2021 daemon.debug pptpd[13165]: CTRL: Reaping child PPP[13166]
Fri May  7 22:31:05 2021 daemon.info pptpd[13165]: CTRL: Client 49.78.205.197 control connection finished

eearphon, 2021-06-14:
最新 17998489cbdcdf290e7596328d7ec6d3ca9911a3 正常

Lienol, 2021-06-28:
测试正常

closed 2021-06-28 by Lienol