Lienol/openwrt #596: 21.02编译好后点击网络-接口报错,望大神帮忙看看是啥问题

trombonist852, 2021-06-07:
[image]
点击继续后还是会跳这个界面

li-xunhuan, 2021-06-10:
官方新版本已经修复了，等大佬合并代码吧~~

trombonist852, 2021-06-10:
> 官方新版本已经修复了，等大佬合并代码吧~~

好的!期待!哈哈

li-xunhuan, 2021-06-10:
> 好的!期待!哈哈

给你个临时解决方案，编辑`feeds/luci/modules/luci-mod-network/htdocs/luci-static/resources/view/network/interfaces.js`文件的第`362`行，把`uci.add`改成`uci.callSet`，然后重新编译就可以了。

li-xunhuan, 2021-06-10:
[image]

trombonist852, 2021-06-10:
> > 好的!期待!哈哈
> 
> 给你个临时解决方案，编辑`feeds/luci/modules/luci-mod-network/htdocs/luci-static/resources/view/network/interfaces.js`文件的第`362`行，把`uci.add`改成`uci.callSet`，然后重新编译就可以了。

感谢大神!!!!!我一会儿去试试!!!太感谢啦!

Lienol, 2021-06-20:
以后就用master吧，21.02有点上不上 下不下的感觉了。先drop掉21.02分支了。

closed 2021-06-20 by Lienol