Lienol/openwrt #598: Transmission 一添加下载任务就崩溃

eearphon, 2021-06-15:
最新版 1799848 ，默认编译 x86-64
`Tue Jun 15 08:53:35 2021 user.notice transmission: Starting with 3999232000 virt mem
Tue Jun 15 08:53:35 2021 daemon.warn procd: Seccomp support for transmission::instance1 not available
Tue Jun 15 10:16:25 2021 kern.info kernel: [ 4996.993848] traps: transmission-da[5213] general protection ip:7efcefee78f7 sp:7efcefe16320 error:0 in libwolfssl.so.24.4.0[7efcefe8d000+7f000]
Tue Jun 15 10:16:30 2021 daemon.warn procd: Seccomp support for transmission::instance1 not available
Tue Jun 15 10:20:25 2021 kern.info kernel: [ 5236.939057] traps: transmission-da[10868] general protection ip:7f331eaedbea sp:7f331e87e760 error:0 in libc.so[7f331eaeb000+49000]
Tue Jun 15 10:20:30 2021 daemon.warn procd: Seccomp support for transmission::instance1 not available
Tue Jun 15 10:20:57 2021 kern.info kernel: [ 5268.809090] transmission-da[10961]: segfault at 10 ip 00007fdeeff4f3f9 sp 00007fdeefcde9e0 error 6 in libc.so[7fdeeff4c000+49000]
Tue Jun 15 10:21:02 2021 daemon.warn procd: Seccomp support for transmission::instance1 not available
Tue Jun 15 10:21:29 2021 kern.info kernel: [ 5300.680062] transmission-da[10978]: segfault at 7f4112b31998 ip 00007f41129cc670 sp 00007f41128e6fe0 error 7 in libwolfssl.so.24.4.0[7f411295e000+7f000]
Tue Jun 15 10:21:34 2021 daemon.warn procd: Seccomp support for transmission::instance1 not available
Tue Jun 15 10:21:58 2021 kern.info kernel: [ 5329.615240] transmission-da[11005]: segfault at 7f465d48a098 ip 00007f465d35f670 sp 00007f465d27a1e0 error 7 in libwolfssl.so.24.4.0[7f465d2f1000+7f000]
Tue Jun 15 10:22:03 2021 daemon.warn procd: Seccomp support for transmission::instance1 not available
Tue Jun 15 10:22:12 2021 kern.info kernel: [ 5343.120989] transmission-da[11247]: segfault at 7fc3b0177c38 ip 00007fc3b0012670 sp 00007fc3aff2cfe0 error 7 in libwolfssl.so.24.4.0[7fc3affa4000+7f000]
Tue Jun 15 10:22:17 2021 daemon.warn procd: Seccomp support for transmission::instance1 not available
Tue Jun 15 10:22:35 2021 daemon.info procd: Instance transmission::instance1 s in a crash loop 6 crashes, 19 seconds since last crash`

eearphon, 2021-06-16:
用 https://github.com/Lienol/openwrt-packages main 分支的 transmission-daemon - 3.00-11 替换，恢复正常！

Lienol, 2021-06-20:
已更新

closed 2021-06-20 by Lienol

eearphon, 2021-07-01:
> 已更新

 transmission-daemon - 3.00 ，重启机器后，下载任务就丢失