Lienol/openwrt #605: 19.07分支编译x86失败，望修复，感谢！

281677160, 2021-06-27:
install -d -m0755 /workdir/openwrt/bin/packages/x86_64/luci
/workdir/openwrt/scripts/ipkg-build -m "" /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-ttyd/ipkg-all/luci-app-ttyd /workdir/openwrt/bin/packages/x86_64/luci
Packaged contents of /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-ttyd/ipkg-all/luci-app-ttyd into /workdir/openwrt/bin/packages/x86_64/luci/luci-app-ttyd_1.0-3_all.ipk
mkdir -p /workdir/openwrt/bin/targets/x86/64/packages /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-ttyd/ipkg-all/luci-i18n-ttyd-zh-cn/CONTROL /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-ttyd/ipkg-all/luci-i18n-ttyd-zh-cn/etc/uci-defaults
echo "uci set luci.languages.zh_cn='中文 (Chinese)'; uci commit luci" > /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-ttyd/ipkg-all/luci-i18n-ttyd-zh-cn/etc/uci-defaults/luci-i18n-ttyd-zh-cn
install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-ttyd/ipkg-all/luci-i18n-ttyd-zh-cn/usr/lib/lua/luci/i18n
po2lmo /workdir/openwrt/feeds/luci/transplant/luci-app-ttyd/po/zh-cn/terminal.po /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-ttyd/ipkg-all/luci-i18n-ttyd-zh-cn/usr/lib/lua/luci/i18n/terminal.zh-cn.lmo;
find /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-ttyd/ipkg-all/luci-i18n-ttyd-zh-cn -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/workdir/openwrt/staging_dir/host/bin/sstrip" STRIP_KMOD="/workdir/openwrt/scripts/strip-kmod.sh" PATCHELF="/workdir/openwrt/staging_dir/host/bin/patchelf" /workdir/openwrt/scripts/rstrip.sh /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-ttyd/ipkg-all/luci-i18n-ttyd-zh-cn
(cd /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-ttyd/ipkg-all/luci-i18n-ttyd-zh-cn/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm; echo "$V_Package_luci_i18n_ttyd_zh_cn_postinst" > postinst-pkg; chmod 0755 postinst-pkg; )
install -d -m0755 /workdir/openwrt/bin/packages/x86_64/luci
/workdir/openwrt/scripts/ipkg-build -m "" /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-ttyd/ipkg-all/luci-i18n-ttyd-zh-cn /workdir/openwrt/bin/packages/x86_64/luci
Packaged contents of /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-ttyd/ipkg-all/luci-i18n-ttyd-zh-cn into /workdir/openwrt/bin/packages/x86_64/luci/luci-i18n-ttyd-zh-cn_1.0-3_all.ipk
echo "luci-app-ttyd" >> /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/luci-app-ttyd.default.install
rm -rf /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-ttyd/.pkgdir/luci-i18n-ttyd-zh-cn.installed /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-ttyd/.pkgdir/luci-i18n-ttyd-zh-cn
mkdir -p /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-ttyd/.pkgdir/luci-i18n-ttyd-zh-cn
install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-ttyd/.pkgdir/luci-i18n-ttyd-zh-cn/etc/uci-defaults
echo "uci set luci.languages.zh_cn='中文 (Chinese)'; uci commit luci" > /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-ttyd/.pkgdir/luci-i18n-ttyd-zh-cn/etc/uci-defaults/luci-i18n-ttyd-zh-cn
install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-ttyd/.pkgdir/luci-i18n-ttyd-zh-cn/usr/lib/lua/luci/i18n
po2lmo /workdir/openwrt/feeds/luci/transplant/luci-app-ttyd/po/zh-cn/terminal.po /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-ttyd/.pkgdir/luci-i18n-ttyd-zh-cn/usr/lib/lua/luci/i18n/terminal.zh-cn.lmo;
touch /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-ttyd/.pkgdir/luci-i18n-ttyd-zh-cn.installed
mkdir -p /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-ttyd/.pkgdir/luci-i18n-ttyd-zh-cn/. /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.luci-i18n-ttyd-zh-cn_installed
echo "luci-i18n-ttyd-zh-cn" >> /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/luci-app-ttyd.default.install
make[3]: Leaving directory '/workdir/openwrt/feeds/luci/transplant/luci-app-ttyd'
time: package/feeds/luci/luci-app-ttyd/compile#0.34#0.11#0.42
make[3]: Entering directory '/workdir/openwrt/feeds/packages/libs/libcap-ng'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.libcap-ng-0.7.10.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "libcap-ng-0.7.10.tar.gz" "a84ca7b4e0444283ed269b7a29f5b6187f647c82e2b876636b49b9a744f0ffbf" "" "https://people.redhat.com/sgrubb/libcap-ng"    '
mkdir -p /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/libcap-ng-0.7.10/.pkgdir/libcap-ng/. /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.libcap-ng_installed
echo "libcap-ng" >> /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/libcap-ng.default.install
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/libcap-ng'
time: package/feeds/packages/libcap-ng/compile#0.18#0.04#0.21
make[3]: Entering directory '/workdir/openwrt/package/libs/libmnl'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.libmnl-1.0.4.tar.bz2.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "libmnl-1.0.4.tar.bz2" "171f89699f286a5854b72b91d06e8f8e3683064c5901fb09d954a9ab6f551f81" "" "http://www.netfilter.org/projects/libmnl/files" "ftp://ftp.netfilter.org/pub/libmnl"    '
mkdir -p /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
echo '0' | cmp -s - /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/libmnl.version || { echo '0' > /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/libmnl.version;  } 
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/libmnl-1.0.4/.pkgdir/libmnl/. /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.libmnl_installed
echo "libmnl" >> /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/libmnl.default.install
make[3]: Leaving directory '/workdir/openwrt/package/libs/libmnl'
time: package/libs/libmnl/compile#0.25#0.06#0.30
make[3]: Entering directory '/workdir/openwrt/package/libs/libnfnetlink'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.libnfnetlink-1.0.1.tar.bz2.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "libnfnetlink-1.0.1.tar.bz2" "f270e19de9127642d2a11589ef2ec97ef90a649a74f56cf9a96306b04817b51a" "" "http://www.netfilter.org/projects/libnfnetlink/files/" "ftp://ftp.netfilter.org/pub/libnfnetlink/"    '
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/libnfnetlink-1.0.1/.configured_*
rm -f /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.libnfnetlink_installed
(cd /workdir/openwrt/build_dir/target-x86_64_musl/libnfnetlink-1.0.1; rm -f aclocal.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf autom4te.cache; [ -e ./config.rpath ] || ln -s /workdir/openwrt/scripts/config.rpath ./config.rpath; touch NEWS AUTHORS COPYING ABOUT-NLS ChangeLog; AUTOM4TE=/workdir/openwrt/staging_dir/host/bin/autom4te AUTOCONF=/workdir/openwrt/staging_dir/host/bin/autoconf AUTOMAKE=/workdir/openwrt/staging_dir/host/bin/automake ACLOCAL=/workdir/openwrt/staging_dir/host/bin/aclocal AUTOHEADER=/workdir/openwrt/staging_dir/host/bin/autoheader LIBTOOLIZE=/workdir/openwrt/staging_dir/host/bin/libtoolize LIBTOOL=/workdir/openwrt/staging_dir/host/bin/libtool M4=/workdir/openwrt/staging_dir/host/bin/m4 AUTOPOINT=true LIBTOOLIZE='/workdir/openwrt/staging_dir/host/bin/libtoolize --install' /workdir/openwrt/staging_dir/host/bin/autoreconf -v -f -i -s -B /workdir/ [... cut ...]
make[3]: *** [Makefile:76: /workdir/openwrt/build_dir/target-x86_64_musl/libnfnetlink-1.0.1/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 127
make[3]: Leaving directory '/workdir/openwrt/package/libs/libnfnetlink'
time: package/libs/libnfnetlink/compile#4.39#0.38#5.68
make[2]: Leaving directory '/workdir/openwrt'
make[2]: *** [package/Makefile:115: package/libs/libnfnetlink/compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:109: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make: *** [/workdir/openwrt/include/toplevel.mk:223: world] Error 2

Lienol, 2021-06-27:
已修复

closed 2021-06-27 by Lienol

281677160, 2021-06-27:
感谢！感谢！