Lienol/openwrt #606: 编译报错 向大佬报告

Saber9361, 2021-06-29:
mkdir -p /home/kioku/openwrt/bin/targets/x86/64/packages /home/kioku/openwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.06/ipkg-x86_64/grub2-bios-setup/CONTROL /home/kioku/openwrt/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/kioku/openwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.06/ipkg-x86_64/grub2-bios-setup/usr/sbin
install -m0755 /home/kioku/openwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.06/grub-bios-setup /home/kioku/openwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.06/ipkg-x86_64/grub2-bios-setup/usr/sbin/
find /home/kioku/openwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.06/ipkg-x86_64/grub2-bios-setup -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/home/kioku/openwrt/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/kioku/openwrt/scripts/strip-kmod.sh" PATCHELF="/home/kioku/openwrt/staging_dir/host/bin/patchelf" /home/kioku/openwrt/scripts/rstrip.sh /home/kioku/openwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.06/ipkg-x86_64/grub2-bios-setup
rstrip.sh: /home/kioku/openwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.06/ipkg-x86_64/grub2-bios-setup/usr/sbin/grub-bios-setup: executable
(cd /home/kioku/openwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.06/ipkg-x86_64/grub2-bios-setup/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/kioku/openwrt/bin/targets/x86/64/packages
/home/kioku/openwrt/staging_dir/host/bin/fakeroot /home/kioku/openwrt/scripts/ipkg-build -m "" /home/kioku/openwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.06/ipkg-x86_64/grub2-bios-setup /home/kioku/openwrt/bin/targets/x86/64/packages
Packaged contents of /home/kioku/openwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.06/ipkg-x86_64/grub2-bios-setup into /home/kioku/openwrt/bin/targets/x86/64/packages/grub2-bios-setup_2.06-1_x86_64.ipk
rm -rf /home/kioku/openwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.06/.pkgdir/grub2-bios-setup.installed /home/kioku/openwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.06/.pkgdir/grub2-bios-setup
mkdir -p /home/kioku/openwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.06/.pkgdir/grub2-bios-setup
install -d -m0755 /home/kioku/openwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.06/.pkgdir/grub2-bios-setup/usr/sbin
install -m0755 /home/kioku/openwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.06/grub-bios-setup /home/kioku/openwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.06/.pkgdir/grub2-bios-setup/usr/sbin/
touch /home/kioku/openwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.06/.pkgdir/grub2-bios-setup.installed
mkdir -p /home/kioku/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/kioku/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/kioku/openwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.06/.pkgdir/grub2-bios-setup/. /home/kioku/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/kioku/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.grub2-bios-setup_installed
echo "grub2-bios-setup" >> /home/kioku/openwrt/staging_dir/target-x86_64_musl/pkginfo/grub2.none.install
make[3]: Leaving directory '/home/kioku/openwrt/package/boot/grub2'
time: package/boot/grub2/none/compile#43.66#5.79#54.34
make[2]: Leaving directory '/home/kioku/openwrt'
make[1]: *** [package/Makefile:110: /home/kioku/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/kioku/openwrt'
make: *** [/home/kioku/openwrt/include/toplevel.mk:230: world] Error 2

2286927, 2021-06-30:
和你的不太一样
ucm_exec.c:177:11: error: 'PATH_MAX' undeclared (first use in this function); did you mean 'INT8_MAX'?
  char bin[PATH_MAX];
           ^~~~~~~~
           INT8_MAX
make[6]: *** [ucm_exec.lo] Error 1
Makefile:379: recipe for target 'ucm_exec.lo' failed
make[6]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/alsa-lib-1.2.5/src/ucm'
make[5]: *** [all-recursive] Error 1
Makefile:540: recipe for target 'all-recursive' failed
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/alsa-lib-1.2.5/src'
make[4]: *** [all-recursive] Error 1
Makefile:403: recipe for target 'all-recursive' failed
make[3]: *** [/workdir/openwrt/build_dir/target-x86_64_musl/alsa-lib-1.2.5/.built] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/alsa-lib-1.2.5'
Makefile:120: recipe for target '/workdir/openwrt/build_dir/target-x86_64_musl/alsa-lib-1.2.5/.built' failed
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/alsa-lib'
time: package/feeds/packages/alsa-lib/compile#0.33#0.08#0.40
    ERROR: package/feeds/packages/alsa-lib failed to build.
package/Makefile:114: recipe for target 'package/feeds/packages/alsa-lib/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
make[2]: *** [package/feeds/packages/alsa-lib/compile] Error 1
make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
package/Makefile:110: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make: *** [world] Error 2
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:228: recipe for target 'world' failed

kl9800, 2021-06-30:
我也是这个问题，一模一样，
全新的乌班图20系统，
sudo apt-get update这些照做的，
期间没有发生任何问题，
源码方面，从git pull重头来的，没有任何修改，
也这样，求大佬解决，
dl下的包也齐全，需要下载的全下来了。

2286927, 2021-07-01:
我仔细看了下分别出现了这些问题
 ERROR: package/feeds/packages/alsa-lib failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [world] Error 1
/workdir/openwrt/include/toplevel.mk:228: recipe for target 'world' failed
WARNING: Makefile 'package/openwrt-packages/naiveproxy/Makefile' has a build dependency on 'ninja/host', which does not exist

2286927, 2021-07-01:
试试添加“config-5.4”文件（该文件在”https://github.com/Lienol/openwrt/tree/main/target/linux/x86/64“里面获取），然后在yml文件里添加“KERNEL_CONF: config-5.4”如下

env:
  REPO_URL: https://github.com/Lienol/openwrt
  REPO_BRANCH: main
  FEEDS_CONF: feeds.conf.default
  KERNEL_CONF: config-5.4

2286927, 2021-07-01:
试试添加“config-5.4”文件（该文件在”[](https://github.com/Lienol/openwrt/tree/main/target/linux/x86/64)“里面获取），然后在yml文件里添加“KERNEL_CONF: config-5.4”如下

env:
REPO_URL: https://github.com/Lienol/openwrt
REPO_BRANCH: main
FEEDS_CONF: feeds.conf.default
KERNEL_CONF: config-5.4

2286927, 2021-07-01:
试试添加“config-5.4”文件（该文件在”[地址](https://github.com/Lienol/openwrt/tree/main/target/linux/x86/64)“里面获取），然后在yml文件里添加“KERNEL_CONF: config-5.4”如下

env:
REPO_URL: https://github.com/Lienol/openwrt
REPO_BRANCH: main
FEEDS_CONF: feeds.conf.default
KERNEL_CONF: config-5.4

kl9800, 2021-07-03:
谢谢大佬！

Saber9361, 2021-07-03:
感谢 已解决

closed 2021-07-03 by Saber9361