Lienol/openwrt #616: 20210711及0712编译mwan3重启不生效

imace3, 2021-07-12:
mwan3重启后不启动，策略分流失效。
需要手动重新保存更新其中一个wan口，才能让wan口上线，策略分流生效。

midorima5791, 2021-07-18:
are you using passwall loadbalacing?

imace3, 2021-07-18:
yes. I am using passwall loadbalacing.

midorima5791, 2021-07-18:
same issue with me not working mwan3 using passwall not working both server

imace3, 2021-07-18:
update mwan3 to 2.10.11-1 and luci-app-mwan3 to git-21.190.23891-f5f6b3e, it's working but have to restart the mwan3 service every time when openwrt reboots.

midorima5791, 2021-07-18:
are you using 19.07 openwrt? version?

imace3, 2021-07-18:
OpenWrt SNAPSHOT / LuCI Master Mod by Lienol git-8b82e4c

I just tried not to use passwall's loadbalancing. and then reboot. mwan3 still not working.

check the mwan3 logs, it it said error (15). not sure what the problem is.

imace3, 2021-07-27:
mwan3 | 2.10.11-1
luci-app-mwan3 | git-21.190.23891-f5f6b3e
7月27日最新版本也都还是重启后不生效。

closed 2025-06-12 by Lienol