Lienol/openwrt #658: ERROR: package/wwan/driver/quectel-qmi-wwan failed to build.

intannajwa, 2025-06-27:
### Describe the bug

 make[3] -C package/wwan/driver/quectel-qmi-wwan compile
    ERROR: package/wwan/driver/quectel-qmi-wwan failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/runner/work/openwrt/openwrt/include/toplevel.mk:233: world] Error 1
Error: Process completed with exit code 2.

### OpenWrt version

24.10

### OpenWrt release

24.10

### OpenWrt target/subtarget

24.10

### Device

aw1000

### Image kind

Official downloaded image

### Diffconfig

```text
ERROR: package/wwan/driver/quectel-qmi-wwan failed to build.
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Lienol, 2025-06-28:
`Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on`

Lienol, 2025-06-28:
[image]
The test went fine.

intannajwa, 2025-06-29:
> `Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on`

here is full log https://raw.githubusercontent.com/intannajwa/CONFIG/refs/heads/main/Lienol/0_build.txt

Lienol, 2025-06-29:
Please re-pull the code and compile.

closed 2025-08-03 by Lienol